GabrielUtt/StrayCat_HomeTemp #2: Recherche Virtualisierung Arduino

GabrielUtt, 2025-01-05:
### **Recherche Virtualisierung Arduino**

---

### **Beschreibung**

- **Warum:** 
- **Was:** 

---

### **Abhängigkeiten**

- Abhängigkeit 1  
- Abhängigkeit 2  

---

### **Ressourcen/Referenzen**

- [Referenz 1](#)  
- [Referenz 2](#)  

---

### **Zuständige und Labels**
- **Zuständige:** 
- **Labels:** 

---

### **Akzeptanzkriterien**

- [ ] Kriterium 1  
- [ ] Kriterium 2  
- [ ] Kriterium 3

kaptenblaubar, 2025-01-09:
[image]

kaptenblaubar, 2025-01-09:
https://www.tinkercad.com/things/50s3SQYFf7F-editing-components/editel?returnTo=https%3A%2F%2Fwww.tinkercad.com%2Fdashboard

kaptenblaubar, 2025-01-09:
simple ampel

closed 2025-01-09 by kaptenblaubar

GabrielUtt, 2025-01-09:
Link funktioniert nicht

GabrielUtt, 2025-01-09:
Info für zukünftige Schaltpläne in Arduino: 

In Arduino-Schaltplänen gibt es typische Farbcodierungen für die verschiedenen Verbindungen. Diese sind jedoch nicht strikt standardisiert, sondern basieren auf gängigen Konventionen, die Klarheit und Lesbarkeit fördern. Hier sind die üblichen Farben:

Farbcodierung:
Power (z. B. 5V oder 3.3V):

Typische Farbe: Rot
Begründung: Rot wird oft für positive Spannungsleitungen verwendet, um die Stromversorgung hervorzuheben.
Ground (GND):

Typische Farbe: Schwarz
Begründung: Schwarz steht traditionell für Masse oder Rückleitung in Gleichstromsystemen.
Signal:

Typische Farben: Gelb, Grün oder Blau
Begründung: Diese Farben werden oft für Daten- oder Steuerleitungen verwendet, um sie von Strom- und Masseleitungen abzugrenzen.

GabrielUtt, 2025-02-25:
www.wokwi.com bietet wesentlich mehr Möglichkeiten für Arduino Simulationen, anbei der Nachbau eines Tutorials für Temperatursensor (Board + Code)

https://wokwi.com/projects/421628414510088193