GabrielUtt/StrayCat_HomeTemp #18: Temperatur Zeit auf SD Karte speichern

kaptenblaubar, 2025-04-14:
Ein kleines Programm bei dem in kurzen abständen die zeit mit der aktuellen Temperatur auf die SD-Karte gespeichert wird.

Physische und digitale umsetzung

kaptenblaubar, 2025-04-14:
https://wokwi.com/projects/428203200077432833 Virtuelle Umsetzung des Programms physische testwerte und Code in C:StrayCat_HomeTemp\virtual_arduino\Saving_Temp+Time_on_SDCard\Temp_Time_combined_SDCard

GabrielUtt, 2025-04-14:
virtuelle Umsetzung für Dokumentation, funktioniert in Wokwi allerdings nicht, möglicherweise wegen unterschiedlicher Bauteile

GabrielUtt, 2025-04-21:
Upgedatete Version findet sich unter Programm_Temp_Zeit_SD_Karte_V2.0 
..\GitHub\StrayCat_HomeTemp\virtual_arduino

Änderungen: 
Code neu strukturiert, pro Tag werden einzelne Dateien abgespeichert

closed 2025-05-06 by GabrielUtt